ros-planning/navigation_experimental #25: sbpl_lattice_planner causes move_base to crash upon map resize

li9i, 2018-10-02:
My setup is as follows:

I am running a turtlebot in Gazebo under kinetic. I am using gmapping to map the willowgarage world, and I am using the sbpl_lattice_planner as my global_planner. Everything runs smoothly at first: I provide navigation goals through rviz and, up until the map expands, the robot navigates successfully to them. However, after the map is resized, upon providing the next goal, move_base crashes with the following message:

```
(init info etc)
[ INFO] [1538471591.487866828, 0.705000000]: Resizing costmap to 1088 X 1248 at 0.025000 m/pix
[ INFO] [1538471591.592609722, 0.805000000]: Received a 1088 X 1248 map at 0.025000 m/pix
[...]
[execution msgs]
[...]
[ INFO] [1538401212.258390982, 40.253000000]: Resizing costmap to 1504 X 1248 at 0.025000 m/pix
Laser Pose= -2.72069 2.90614 2.51478
[ INFO] [1538401212.585336613, 40.555000000]: [sbpl_lattice_planner] getting start point (-2.62449,2.84664) goal point (-2.66432,2.93017)
[move_base-11] process has died [pid 5492, exit code -11, cmd /opt/ros/kinetic/lib/move_base/move_base cmd_vel:=navigation_velocity_smoother/raw_cmd_vel odom:=odom scan:=scan __name:=move_base __log:=/home/li9i/.ros/log/697d1ec6-c57f-11e8-af3c-54e6fc90f68a/move_base-11.log].
```

When I use navfn as the global planner, the crash does not occur, hence I think that the lattice planner is the culprit. I can provide further details upon request.

li9i, 2018-10-02:
Sure, of course. Here is a rosbag info of the [attached](https://github.com/li9i/rosbags/blob/master/2018-10-02-16-48-55.bag):

```
$ rosbag info 2018-10-02-16-48-55.bag 
path:        2018-10-02-16-48-55.bag
version:     2.0
duration:    47.8s
start:       Jan 01 1970 02:00:08.60 (8.60)
end:         Jan 01 1970 02:00:56.37 (56.37)
size:        15.9 MB
messages:    50135
compression: none [15/15 chunks]
types:       geometry_msgs/PoseStamped         [d3812c3cbc69362b77dc0b19b345f8f5]
             move_base_msgs/MoveBaseActionGoal [660d6895a1b9a16dce51fbdd9a64a56b]
             nav_msgs/OccupancyGrid            [3381f2d731d4076ec5c71b0759edbe4e]
             nav_msgs/Path                     [6227e2b7e9cce15051f669a5e197bbf7]
             tf2_msgs/TFMessage                [94810edda583a504dfda3829e70d7eec]
topics:      /map                                     6 msgs    : nav_msgs/OccupancyGrid           
             /move_base/SBPLLatticePlanner/plan       4 msgs    : nav_msgs/Path                    
             /move_base/current_goal                  5 msgs    : geometry_msgs/PoseStamped        
             /move_base/goal                          5 msgs    : move_base_msgs/MoveBaseActionGoal
             /tf                                  50114 msgs    : tf2_msgs/TFMessage                (3 connections)
             /tf_static                               1 msg     : tf2_msgs/TFMessage

```

I have included tf and the planner's plan for convenience if you want to visualise the situation.

mintar, 2018-10-10:
Thanks! This is fixed now.

li9i, 2018-10-10:
I can confirm that it's fixed. My sincere thanks.