ros-planning/navigation_experimental #32: ERROR: failed to read in motion primitive file

thanhthtran, 2018-12-23:
Hi, i would like to try the spbl_laptice on mir 100 robot. However, it shows the error  ERROR: failed to read in motion primitive file. would you know how i could fix it please?

mintar, 2019-01-16:
One more comment (for later reference): This error means that the motion primitive file (*.mprim) was found and read, but SBPL didn't like something about it. Usually, this means that the global costmap doesn't have the same resolution as the motion primitive file.

closed 2019-01-16 by mintar

Combinacijus, 2020-08-11:
@mintar Would it be possible output more detailed exception? Instead of generic "ERROR: failed to read in motion primitive file" (which is not actually true as you said it was read) to something like "ERROR: primitive file and global map resolutions are different 0.025 and 0.05 respectively"?

mintar, 2020-08-14:
That error comes from the underlying SBPL library, so I cannot control the output. However, if you clone my fork of sbpl, check out the "enable_roslog" branch, compile and install it, you'll get much better error output from SBPL:

https://github.com/mintar/sbpl/tree/enable_roslog