ros-planning/navigation_experimental #34: pose_follower generates y-velocity

veeraragav, 2019-02-05:
I set the holonomic to false using rqt_reconfigure. Still it produces velocities in y direction. All other parameters are default.

mintar, 2019-02-05:
The pose_follower local planner uses two sets of parameters:

1. Its own parameters, among them `holonomic`. These are not dynamically reconfigurable.
2. The parameters from the TrajectoryPlannerROS instance that is used internally, among them `holonomic_robot`. Those are dynamically reconfigurable.

You probably set `holonomic_robot` to false, but `holonomic` was still true. You should set `holonomic` to false in the launch file (you can't use rqt_reconfigure for that).

I hope this helps.

closed 2019-02-05 by mintar

veeraragav, 2019-02-05:
Thank You. I changed the default value to false.

tomin11, 2019-10-02:
@veeraragav  did setting `holonomic` to false help ? It did not help me. I still get y velocities. ( of around 0.9 m/s)

mintar, 2019-10-10:
If `holonomic` is set to false, pose_follower shouldn't generate y velocities. You've probably set the parameter in the wrong namespace. I find [rosautodoc](https://github.com/bponsler/rosautodoc) quite useful to figure out what parameters a node *actually* reads. Compare this to the output of `rosparam dump`, and you'll probably see where the mistake is.