ros-planning/navigation_experimental #35: pose_follower's global plan transformation issue

veeraragav, 2019-02-05:
My local costmap or local planner's reference frame is `/odom`. When the pose_follower receives the global plan through `setPlan()`, it converts the plan's coordinates from `\map` frame to `\odom' frame.
I use AMCL for localization, which makes the `/odom` frame move away from `/map` frame. But the pose_follower doesn't reconvert the global plan to `/odom` frame. So the robot doesn't reach the goal specified in the `/map` frame. I took a snapshot of rviz after the robot had reached the goal. The global plan is the green line along the `/map` frame's x axis. The robot had reached the goal with respect to `/odom` frame, but the actual goal is somewhere else because of the movement between `/map` and `/odom`. How do I address this problem?
[image]

mintar, 2019-02-05:
Just so I understand correctly: When you gave the goal, the odom frame was at the map frame? And during execution, it drifted away as far as shown in the image? In that case, that is the problem. It shouldn't drift away so quickly.

veeraragav, 2019-02-05:
You are right.

veeraragav, 2019-02-05:
But AMCL tries to compensate the error in wheel odometry by shifting the map frame so that the robot is properly localized.

veeraragav, 2019-02-05:
The wheel odometry is very bad in my case. I am also trying to make it work with the global frame of local planner set to /map. In this case the robot starts rotating as soon as i give the goal.

mintar, 2019-02-05:
I don't think this is a bug in pose_follower, so I'm going to close this issue. Ok?

veeraragav, 2019-02-05:
It would be great if you transform the plan for each computeVelocityCommands() function call. That's how DWA planner works.

mintar, 2019-02-06:
I guess you are right, maybe the plan should be transformed every time to guarantee that the robot ends up at the goal in the map frame.

I'm afraid though that this will be equivalent to setting the local planner's global frame to /map, so in your case, the robot will start rotating (because relative to /map, the robot's position has just jumped away from the global plan, so it's trying to get back to the global plan). Either way, there's no way around fixing the odometry/localization problem of your robot.

veeraragav, 2019-02-12:
Okay.

closed 2019-02-12 by veeraragav