ros-planning/navigation_experimental #37: SBPL_Exception  on  kinetic

GITSHOHOKU, 2019-04-11:
I start use sbpl on ros-kinetic, I use  sudo apt-get install ros-kinetic-sbpl, sbpl-lattice-planner 
I set it as the base_global_planner in launch file like this:
`<node pkg="move_base" type="move_base" respawn="false" name="move_base" clear_params="true" output="screen">
    <param name="footprint_padding" value="0.01" />
    <param name="controller_frequency" value="10.0" />
    <param name="controller_patience" value="100.0" />
    <param name="base_global_planner" value="SBPLLatticePlanner" />
    <param name="base_local_planner" value="pose_follower/PoseFollower" />

    <rosparam file="$(find sx_robot)/config/sx_params/costmap_common_params.yaml" command="load" ns="global_costmap" />
    <rosparam file="$(find sx_robot)/config/sx_params/costmap_common_params.yaml" command="load" ns="local_costmap" />
    <rosparam file="$(find sx_robot)/config/sx_params/local_costmap_params.yaml" command="load" />
    <rosparam file="$(find sx_robot)/config/sx_params/global_costmap_params.yaml" command="load" />
    
    <rosparam file="$(find sx_robot)/config/sx_params/pose_follower_params.yaml" command="load" />
    <rosparam file="$(find sx_robot)/config/sx_params/sbpl_global_params.yaml" command="load" />
  </node>`
But when I launched it ,I got  these error message:
`[INFO] [1554984329.226987]: simulate startup!
[ INFO] [1554984329.564472698]: Using plugin "static_layer"
[ INFO] [1554984329.569651035]: Requesting the map...
[ INFO] [1554984329.772388905]: Resizing costmap to 600 X 600 at 0.010000 m/pix
[ INFO] [1554984329.872164369]: Received a 600 X 600 map at 0.010000 m/pix
[ INFO] [1554984329.878223848]: Using plugin "obstacle_layer"
[ INFO] [1554984329.880451677]:     Subscribed to Topics: realsense_fake_laser
[ INFO] [1554984329.903446263]: Using plugin "inflation_layer"
[ INFO] [1554984329.952730559]: Name is SBPLLatticePlanner
terminate called after throwing an instance of 'SBPL_Exception'
  what():  ERROR: unable to open 
[move_base-7] process has died [pid 4208, exit code -6, cmd /opt/ros/kinetic/lib/move_base/move_base __name:=move_base __log:=/home/hkwal/.ros/log/1860a58e-5c52-11e9-a7f8-f8633f5c04b3/move_base-7.log].
log file: /home/hkwal/.ros/log/1860a58e-5c52-11e9-a7f8-f8633f5c04b3/move_base-7*.log
`
What did I miss?
Thanks a lot!

mintar, 2019-04-11:
You forgot to set the parameter `primitive_filename`.

See here:

https://github.com/ros-planning/navigation_experimental/blob/004b5b6a2adc59c1c1af30d76d0cacfa77aa6aff/sbpl_lattice_planner/launch/move_base/move_base_sbpl.xml#L9

And here:

https://github.com/ros-planning/navigation_experimental/tree/kinetic-devel/sbpl_lattice_planner#parameters

GITSHOHOKU, 2019-04-11:
Ok, Thanks,I will try.  Thanks a lot !

closed 2019-05-13 by mintar