ros-planning/navigation_experimental #38: failed to read motion file error not resolved even after changing global costmap resolution

sudo-panda, 2019-05-12:
I am getting the ```ERROR: failed to read motion primitive file``` even though I changed the global costmap resolution to 0.025 (taken from the mprim file). I have read the previous issue that pointed out the error but I do not get what other parameters in the file and my configuration of my bot are conflicting. Could you please help.

mintar, 2019-05-13:
Can you upload the mprim file somewhere? Also, can you do a `rosparam dump params.yaml` and upload that file as well?

sudo-panda, 2019-05-13:
[params.yaml](https://gist.github.com/sudo-panda/0d318f3c9584817d2b66c0d8710d186d)
[pr3.mprim](https://gist.github.com/sudo-panda/1a615ebb223a6e068cbd4a74f72d6b20)

sudo-panda, 2019-05-15:
I am no longer using this planner for my bot hence closing this issue. Thanks for helping.

anish0x2a, 2019-09-30:
I am facing this problem on my car-like robot. I have already used SBPL Lattice Planner with Teb Local Planner on my holonomic bot and it worked perfectly fine. But I can't seem to get past this error now. Is there any different mprim file for car-like robots?

anish0x2a, 2019-10-01:
It was actually my fault. I had missed the delta parameter in the gmapping param file which was essentially the resolution of the map. Matching the map resolution with the mprim resolution worked perfectly. Thanks!

closed 2019-10-04 by mintar