ros-planning/navigation_experimental #39: robot tends to oscillate a lot while moving foward with pose follower local planner

tomin11, 2019-09-26:
Using **navfn as the global planner.** The movement is fine with straight paths and with spin turns. 
But when it comes to curved paths and paths that need the robot to turn first and then move forward, the robot turns to the desired goal position, continues on that path for some time and then it starts oscillating clockwise and anticlockwise after which it continues moving towards the goal. This is repeated till it reaches the goal. This is a diff drive robot.
The pose follower params used are:
PoseFollower:
 k_trans: 2.0
 k_rot: 0.8
 tolerance_trans: 0.15
 tolerance_rot: 0.13
 tolerance_timeout: 0.3
 holonomic: false
 holonomic_robot: false
 samples: 150
 max_vel_lin: 0.5
 min_vel_lin: 0.0
 max_vel_th: 0.20
 min_vel_th: 0.01
 min_in_place_vel_th: 0.0
 in_place_trans_vel: 0.0
 trans_stopped_velocity: 0.0001
 rot_stopped_velocity: 0.0001
 allow_backwards: true
 turn_in_place_first : true
 max_heading_diff_before_moving: 0.3

mintar, 2019-09-26:
PoseFollower is a *really* basic trajectory following algorithm, so consider switching to something more sophisticated. That being said, I've looked at your parameters and the PoseFollower code, and I can't see anything wrong.

Some suggestions you could try:

* Set the PoseFollower log level to "debug", so you can see all the `ROS_DEBUG` messages printed from the code: https://github.com/ros-planning/navigation_experimental/blob/kinetic-devel/pose_follower/src/pose_follower.cpp
* Visualize the current plan and the robot's position in RViz and observe the situations when the oscillations occur. E.g., is the robot parallel to the closest position on the path? Does the odometry pose jump around? Maybe take a video of the screen.
* Ask on https://answers.ros.org/ . :)

tomin11, 2019-09-26:
Thank you for your response. I would like to share a video.
[image]

Here, the white arrow is the odometry and the green line is the global plan of pose follower.

mintar, 2019-09-26:
That is very strange. Could you record a rosbag of the following topics?

* global plan
* odom
* tf
* laser scans
* cmd_vel

tomin11, 2019-09-26:
[pose_follower_issue.tar.gz](https://github.com/ros-planning/navigation_experimental/files/3657509/pose_follower_issue.tar.gz)
Here is the rosbag with the required topics.

tomin11, 2019-10-02:
this was solved by using sbpl as the global planner instead of navfn according to Mr.Eitan in http://ros-users.122217.n3.nabble.com/Nav-Stack-with-nonholonomic-robots-td1404191.html

closed 2019-10-02 by tomin11