ros-planning/navigation_experimental #47: pose_follower: Driving backwards even with allow_backwards=false

nxdefiant, 2020-03-19:
Hi,

the pose_follower is driving backwards with non holonomic robots even when the parameter allow_backwards is set to false.

I think that PoseFollower::limitTwist() is missing a check for this condition, e.g. the following should help:

```
if (!allow_backwards_ && res.linear.x < 0) {
	res.linear.x = 0.0;
}
```

nxdefiant, 2020-03-21:
Btw disabling allow_backwards is a bad idea when using Navfn as a global planner since Navfn can produce paths with all goals in backwards direction.

mintar, 2020-03-23:
Yes, Navfn only produces a sequence of points, not poses (it doesn't fill out the orientation component of the pose).

Simply limiting the twist like you proposed won't help: If pose_follower decided to go backwards, but then sets the velocity to zero, it will just stop in place and not move anywhere.

If you have a working fix, I'll be happy to merge a pull request.