ros-planning/navigation_experimental #51: sbpl_lattice_planner : unable to use motion primitives with 32/64 number of discrete angles

luckydaemon, 2021-04-06:
Not sure if I should ask it here or at spbl main github (probably second, because I am getting an SBPL_exception), but I stack on problem, that I am unable to use sbpl_lattice_planner if  motion primitives generated for "numberofangle" = 32 using genmprim.m script. I saw another issue about hidden special symbols at start of the file, but that is not my case. 

[mprim_example.zip](https://github.com/ros-planning/navigation_experimental/files/6263581/mprim_example.zip)

mintar, 2021-04-12:
If you compile with my SBPL fork (see https://github.com/sbpl/sbpl/pull/56), you get some more output:

```
[ERROR] [1618243328.911517682, 0.900000000]: ERROR: invalid angular resolution 32 angles (instead of 16 angles) in the motion primitives file
 (/move_base_node)
[ERROR] [1618243328.911545461, 0.900000000]: ERROR: failed to read in motion primitive file
```

Grepping for the error, I found this line:

https://github.com/sbpl/sbpl/blob/4d654845aa3b92aa1b4282a342e9011bac95aeb9/src/discrete_space_information/environment_navxythetalat.cpp#L812-L815

However, I did not investigate further (where is this config read?)

mintar, 2021-06-21:
I'm closing this issue due to lack of activity. If you have anything to add, feel free to reopen!

closed 2021-06-21 by mintar