ros-planning/navigation_experimental #53: how to prohibit sbpl global planning through unknown area?

rightandfront, 2021-04-20:
I'm using move_base package in a differ robot with a static global map, which is got from the gmapping. So, there are some unknown area in the map, and I found that sbpl planner can't find a path through those area.
[image]

rightandfront, 2021-04-21:
> I guess you'll have to write a small node that subscribes to the map topic, processes the map the way you want, and re-publishes it on a different topic. You can then use the filtered topic for move_base.
I will have a try. Thanks a lot.

SteveMacenski, 2021-04-21:
@mintar navigation also has a param for tracking unknown costs, I typically use that in the planner to do some logic like `if tracking unknown and cell unknown, don't go` type-logic.

TrayfourYu, 2021-04-23:
Well, I end up this with just clearing out the unknown area of the map in the **Gimap** that is a picture editing software.

closed 2021-04-23 by TrayfourYu