ros-planning/navigation_experimental #54: Question: How to change frame from which global plan is planned?

willemvergouwen, 2021-04-21:
Hi there!

In my current robot implementation, I need the global plan coming from the SBPLLatticePlanner to be planned from a different frame than base_frame. In the previous planner I used this was a parameter, i.e. I could define a different start point for the global plan. Is this something that is possible for the sbpl_lattice_planner?

Thanks!

mintar, 2021-04-23:
Yes, the `makePlan` function accepts an arbitrary start pose:

https://github.com/ros-planning/navigation_experimental/blob/76950deb723f197aa5e706b554a519914025ad20/sbpl_lattice_planner/include/sbpl_lattice_planner/sbpl_lattice_planner.h#L49-L58

Note that the pose has to have the same `frame_id` as the costmap's global frame ID. So perhaps you'll have to use `tf` to transform the pose before passing it to SBPL.

mintar, 2021-04-23:
If I misunderstood you, and all you want to do is change the name of the link on your robot that's used for planning, that has nothing to do with SBPL, but move_base. You'll just have to change the move_base parameter `global_costmap/robot_base_frame`:

https://github.com/ros-planning/navigation/blob/4a3d261daa4e7eafa40bf7e4505f8aa8678d7bd7/move_base/src/move_base.cpp#L72

This works for all planners, including SBPL.

willie013, 2021-04-23:
Thank you for your swift reply! 

I indeed just want to change the name from which the route is planned using SBPL, as my local planner also uses this different frame. 

However, when I change the move_base parameter `global_costmap/robot_base_frame` the route still is planned from the base_frame instead of the frame I set it to be. Any idea what might be happening here?

mintar, 2021-04-23:
Make sure that `move_base` is actually reading the parameter that you have set. I usually do the following in situations like this:

1. Use [rosautodoc](https://github.com/bponsler/rosautodoc/) as a man-in-the-middle to log which parameters are actually read.
2. Use `rosparam dump` to see which parameters are set on the parameter server.
3. Compare

Good luck!

willie013, 2021-04-26:
Thanks for the help mintar! This helps a lot. I will have a look into the debug tools you mentioned, I have never used these tools before (:

willie013, 2021-04-26:
Additionally, is there any way to set the spacing between 2 poses of the planned path using sbpl_lattice_planner? Or do I just have to change the motion primitives to get more spacing between 2 poses of the planned path?

willie013, 2021-04-28:
After going through the source code of move_base_flex I found the issue. By setting the  `robot_frame` parameter, in addition to changing the `global_costmap/robot_base_frame` the route is planned correctly! So I'll close this issue for now, thanks for the help!

closed 2021-04-28 by willie013