ros-planning/navigation_experimental #55: Path planed through the obstacles

TrayfourYu, 2021-04-23:
Hi,
I'm using the sbpl planner on a differ robot. And sometimes I found that the planner produce a path that cross the obstacles when I using a **empty global static map**, but other times it works fine in which the path could just bypass the obstacles. So, I just wonder if the sbpl planner only takes the global costmap into account without any local information when it plans a path.

mintar, 2021-05-04:
All global planners (including) SBPL only take the `global_costmap` into account. But usually you configure your `global_costmap` like [this](https://github.com/dfki-ric/mir_robot/blob/b03d4d2408a6dca863ef500cbd773c88b91a742b/mir_navigation/config/costmap_global_params_plugins_no_virtual_walls.yaml) (so there's a layer for the static map, a layer for the dynamic obstacles and an inflation layer). That way, the dynamic obstacles are also considered by SBPL.

Also make sure that you configure your `inflation_radius` correctly, see https://github.com/ros-planning/navigation_experimental/issues/33#issuecomment-453182448 and https://github.com/ros-planning/navigation_experimental/issues/33#issuecomment-454744475.

mintar, 2021-06-21:
I'm closing this issue due to lack of activity. If you have anything to add, feel free to reopen!

closed 2021-06-21 by mintar