ros-planning/navigation_experimental #57: Slow planning on Jetson TX2

Wimll, 2021-05-11:
Hi there!

I've implemented the SBPL lattice planner for my robot, which works fine in simulation on my computer (AMD Ryzen 7).
However, when I run the algorithm on my robot platform using a Jetson TX2, planning takes ages and only short, straight paths are possible. I've properly set up the motion primitives as I can validate them in simulation.

I came across this issue for SBPL:
https://github.com/sbpl/sbpl/pull/57/commits/bc9a4b326f2dc472bcf3bde554d21c91d2d825e1
https://answers.ros.org/question/194392/problem-running-sbpl-on-arm-odroid-u3/

Is this still a thing for ARM based systems? As the Jetson is ARM based, this would be an explanation for my issues.

The room I am planning in on the robot is larger than in simulation, i.e. ~40*20 m, but this this still should be possible within a adequate timeframe right?

Goal orientations which are 90 degrees shifted from the initial pose don't work and the planner just indicates: "Solution not found". The same holds for navigation goals which are further away than ~7m. Does the lattice planner also return "Solution not found" when the allocated time runs out?

Thanks!

mintar, 2021-05-17:
> Is this still a thing for ARM based systems? As the Jetson is ARM based, this would be an explanation for my issues.

Probably yes. The authors of SBPL sadly haven't maintained it for years. The links that you found seem plausible, so it would probably be a good idea to try the proposed solutions (either compile with `-fsigned-char`, or better switch all `char` to `int8_t` type from stdint.h).

> The room I am planning in on the robot is larger than in simulation, i.e. ~40*20 m, but this this still should be possible within a adequate timeframe right?

I don't have any experience with SBPL on ARM, but I'd say yes.

> Goal orientations which are 90 degrees shifted from the initial pose don't work and the planner just indicates: "Solution not found".

That really looks like a bug, perhaps caused by the "unsigned `char`" problem. If it always fails for some goal orientations, even if the path is short, the reason can't be that SBPL simply ran out of time.

> Does the lattice planner also return "Solution not found" when the allocated time runs out?

Yes.

Wimll, 2021-05-18:
Thanks for your reply @mintar ! Compiling with the `-fsigned-char` flag solved all my issues with running SBPL on ARM. However, it is still quite slow. Do you have any pointers where the code can be optimized? I see you mentioned that a lot of improvements can be made, for instance this pull request and topic:

https://github.com/sbpl/sbpl/pull/40
https://github.com/ros-planning/navigation_experimental/issues/44

But I none of these changes have been pushed upstream, right?

mintar, 2021-05-26:
> But I none of these changes have been pushed upstream, right?

Right. SBPL has basically been unmaintained for years. After sbpl/sbpl#40 was submitted 5 years ago, the authors didn't merge it, so @corot didn't submit any more PRs. #44 tells the whole story, so you already found all relevant info! :)

closed 2021-06-01 by Wimll