ros-planning/navigation_experimental #59: Backward and turn motion primitives

LotfiZ, 2022-05-05:
Hi folks,

I dont know if its relevant to ask this here so please close the issue if not.

I wanted to generate motion primitives using the Matlab script `genmprim_unicycle.m` and i was not able to generate a particular motion that is : backward and turn. I dont know why because i thought its just the opposite of the forward and turn motion, this is my setup : 

```
%0 degreees  
    basemprimendpts0_c = zeros(numberofprimsperangle, 4); %x,y,theta,costmult   
    %x aligned with the heading of the robot, angles are positive  
    %counterclockwise  
    %0 theta change  
    basemprimendpts0_c(1,:) = [1 0 0 forwardcostmult];  
    basemprimendpts0_c(2,:) = [8 0 0 forwardcostmult];  
    basemprimendpts0_c(3,:) = [-1 0 0 backwardcostmult];  
    %1/16 theta change  
    basemprimendpts0_c(4,:) = [8 1 1 forwardandturncostmult];  
    basemprimendpts0_c(5,:) = [8 -1 -1 forwardandturncostmult];  
    %backward and turn  
    basemprimendpts0_c(6,:) = [-8 -1 1 backwardandturncostmult];  
    basemprimendpts0_c(7,:) = [-8 1 -1 backwardandturncostmult];  
```
And i obtained this for 0 degrees : 
[image]

As you can see, forward and turn/backward and turn are not generated the same way. Do you have some idea about this ? 

Thank you !

mintar, 2022-05-06:
That's not the full modified portion of the script. Please post your full modified script.

mintar, 2022-05-06:
You can also have a look at this file: https://github.com/sbpl/sbpl/blob/4d654845aa3b92aa1b4282a342e9011bac95aeb9/matlab/mprim/genmprim_unicycleplussidewaysplusbackturn.m

That file generates backwards turns.

LotfiZ, 2022-05-06:
Hi Martin, thank you for the quick reply. I did not post the full script because i just added the backward and turn motion into the original script : [https://github.com/sbpl/sbpl/blob/4d654845aa3b92aa1b4282a342e9011bac95aeb9/matlab/mprim/genmprim_unicycle.m](url) .
I had a quick look to the script that you provide and i think that the case why the `genmprim_unicycle.m` does not generate the backward and turn motion correctly is because of this condition (im not sure) :    [https://github.com/sbpl/sbpl/blob/4d654845aa3b92aa1b4282a342e9011bac95aeb9/matlab/mprim/genmprim_unicycle.m#L221](url)
`if l < 0
fprintf(1, 'WARNING: l = %d < 0 -> bad action start/end points\n', l);
l = 0;
end;`

Thank you !